ARMmbed/ci-test-shield #5: Some greentea warnings about summary miscount

screamerbg, 2016-10-14:
```
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: 07770221075469163810FF5C
mbedgt: test suite 'tests-assumptions-analogin' ...................................................... OK in 10.82 sec
        test case: 'AnalogIn - can co-exist with DigitalOut' ......................................... OK in 0.07 sec
        test case: 'AnalogIn - existence of `DEVICE_ANALOGIN` macro' ................................. OK in 0.08 sec
        test case: 'AnalogIn - pin can do DigitalOut' ................................................ OK in 0.07 sec
mbedgt: test case summary: 8 passes, 0 failures
mbedgt: utest test case summary mismatch: utest reported passes and failures miscount!
        reported by utest: passes = 8, failures 0)
        test case result count: passes = 3, failures 0)
```

BlackstoneEngineering, 2016-10-14:
Right, thats because of this 

```
Case cases[] = {
    Case("AnalogIn - existence of `DEVICE_ANALOGIN` macro", Macro_Test,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_0>,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_1>,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_2>,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_3>,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_4>,greentea_failure_handler),
    Case("AnalogIn - pin can do DigitalOut", can_digitalout<MBED_CONF_APP_AIN_5>,greentea_failure_handler),
    Case("AnalogIn - can co-exist with DigitalOut", Can_AnalogIn_and_DigitalOut,greentea_failure_handler),

};
```

Each test needs a unique name, otherwise greentea overwrites them. I believe tests are printed out alphabetically by their description. So instead of all the tests having `"AnalogIn - pin can do DigitalOut"` they each need a unique name. Will fix and push shortly.

BlackstoneEngineering, 2016-10-14:
Change pushed

closed 2016-10-14 by BlackstoneEngineering