ARMmbed/ci-test-shield #8: Add UART Tests

BlackstoneEngineering, 2016-10-20:
## Problem
UART Tests cannot be done by Loopback on D0/D1 because some boards (Nucleo / nRF / etc) only have a single UART and use the same peripheral for the Serial Debug lines used by greentea. Thus is these pins are tied together greentea cannot run. 

Therefore the UART tests should be redone to use the Serial Debug channel instead of UART Loopbacks. 

## Solution
Currently not possible

In the future a change should be made to GreenTea that allows for taking control of the serial line and then handing it back. See here : https://github.com/ARMmbed/greentea/issues/196

jeromecoutant, 2017-03-16:
Hi
As tests don't use UART ARDUINO pin, I think you you remove the UART line in the "What is tested?" paragraph...

BlackstoneEngineering, 2017-03-17:
@jeromecoutant The UART is still testsed implicitly by running the tests, as the test framework communicates using the UART channel connected to the interface circuit.

jeromecoutant, 2017-03-17:
Test framework doesn't communicate with the same UART as the one connected to ARDUINO pins...

BlackstoneEngineering, 2017-03-17:
Correct, but it does still validate that the board has a working UART capability. Its not optimal, but its good enough.