ARMmbed/ci-test-shield #10: Meet tests-api-analogin bug?

ccli8, 2016-11-15:
**SHA**
53d447649b9b2edf012347baa10f3c25bb895220

**Description**
I am testing *ci-test-shield* on *NuMaker-PFM-M453* board and meet a potential bug in `tests-api-analogin`. In `TESTS/API/AnalogIn/AnalogIn.cpp` > `AnalogInput_Test`,
```
    AnalogIn ain(ain_pin);
    BusInOut outputs(dout_pin1,dout_pin2,dout_pin3,dout_pin4,dout_pin5);
    outputs.output();
    int x = 0;
    int y= 0;
    outputs = 1;
    float prev_value = 0; 
```
The `outputs` bus should initialize to 0:
```
    AnalogIn ain(ain_pin);
    BusInOut outputs(dout_pin1,dout_pin2,dout_pin3,dout_pin4,dout_pin5);
    outputs.output();
    int x = 0;
    int y= 0;
    outputs = 0;
    float prev_value = 0; 
```

BlackstoneEngineering, 2016-11-15:
Nice find. Fixed here : https://github.com/ARMmbed/ci-test-shield/commit/6ca1b896edff34cacda2ed84e24455747262f295

That said, the PFM-M453 board will still fail the analog in test with the followin results

```
+--------------------------+------------------+--------------------+--------+--------------------+-------------+
| target                   | platform_name    | test suite         | result | elapsed_time (sec) | copy_method |
+--------------------------+------------------+--------------------+--------+--------------------+-------------+
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | FAIL   | 16.47              | shell       |
+--------------------------+------------------+--------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+--------------------------+------------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| target                   | platform_name    | test suite         | test case             | passed | failed | result | elapsed_time (sec) |
+--------------------------+------------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_0 | 0      | 4      | FAIL   | 0.49               |
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_1 | 0      | 5      | FAIL   | 0.6                |
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_2 | 0      | 5      | FAIL   | 0.59               |
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_3 | 0      | 5      | FAIL   | 0.6                |
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_4 | 0      | 5      | FAIL   | 0.6                |
| NUMAKER_PFM_M453-GCC_ARM | NUMAKER_PFM_M453 | tests-api-analogin | Analog Input on AIN_5 | 0      | 5      | FAIL   | 0.59               |
+--------------------------+------------------+--------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 6 FAIL
mbedgt: completed in 21.16 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error code 1.
[mbed] ERROR: Command "mbedgt --test-spec C:\cygwin\home\ausbla01\mbed\ci-test-shield\BUILD\tests\NUMAKER_PFM_M453\GCC_ARM\test_spec.json -n tests-api-analogin -V" in "C:\cygwin\home\ausbla01\mbed\ci-
test-shield"
```

If you uncomment the printf's in the code it looks like this:
```
[1479239861.67][CONN][INF] found KV pair in stream: {{__testcase_start;Analog Input on AIN_0}}, queued...
[1479239861.68][CONN][RXD] X=0
[1479239861.69][CONN][RXD] outputs=0x1
[1479239861.71][CONN][RXD] prevValue=0.000000
[1479239861.72][CONN][RXD] ain=0.000000
[1479239861.72][CONN][RXD]
[1479239861.73][CONN][RXD] X=1
[1479239861.74][CONN][RXD] outputs=0x3
[1479239861.76][CONN][RXD] prevValue=1.000000
[1479239861.77][CONN][RXD] ain=1.000000
[1479239861.77][CONN][RXD]
[1479239861.88][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239861.89][CONN][RXD] X=2
[1479239861.90][CONN][RXD] outputs=0x7
[1479239861.92][CONN][RXD] prevValue=1.000000
[1479239861.93][CONN][RXD] ain=1.000000
[1479239861.93][CONN][RXD]
[1479239862.04][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239862.05][CONN][RXD] X=3
[1479239862.06][CONN][RXD] outputs=0xf
[1479239862.08][CONN][RXD] prevValue=1.000000
[1479239862.09][CONN][RXD] ain=1.000000
[1479239862.09][CONN][RXD]
[1479239862.20][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239862.21][CONN][RXD] X=4
[1479239862.22][CONN][RXD] outputs=0x1f
[1479239862.24][CONN][RXD] prevValue=1.000000
[1479239862.25][CONN][RXD] ain=1.000000
[1479239862.26][CONN][RXD]
[1479239862.37][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239862.41][CONN][INF] found KV pair in stream: {{__testcase_finish;Analog Input on AIN_0;0;4}}, queued...
[1479239862.51][CONN][RXD] >>> 'Analog Input on AIN_0': 0 passed, 4 failed with reason 'Test Cases Failed'
[1479239862.51][CONN][RXD]
[1479239862.56][CONN][RXD] >>> Running case #2: 'Analog Input on AIN_1'...
[1479239862.60][CONN][INF] found KV pair in stream: {{__testcase_start;Analog Input on AIN_1}}, queued...
[1479239862.61][CONN][RXD] X=0
[1479239862.62][CONN][RXD] outputs=0x1
[1479239862.64][CONN][RXD] prevValue=0.000000
[... 138 lines cut ...]
[1479239866.35][CONN][RXD] X=1
[1479239866.36][CONN][RXD] outputs=0x3
[1479239866.38][CONN][RXD] prevValue=1.000000
[1479239866.40][CONN][RXD] ain=1.000000
[1479239866.40][CONN][RXD]
[1479239866.51][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239866.51][CONN][RXD] X=2
[1479239866.52][CONN][RXD] outputs=0x7
[1479239866.54][CONN][RXD] prevValue=1.000000
[1479239866.56][CONN][RXD] ain=1.000000
[1479239866.56][CONN][RXD]
[1479239866.66][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239866.67][CONN][RXD] X=3
[1479239866.68][CONN][RXD] outputs=0xf
[... cut ...]
[1479239866.71][CONN][RXD] ain=1.000000
[1479239866.72][CONN][RXD]
[1479239866.82][CONN][RXD] :31::FAIL: Analog Input did not increment. Check that you have assigned valid pins in mbed_app.json file
[1479239866.83][CONN][RXD] X=4
[1479239866.84][CONN][RXD] outputs=0x1f
[1479239866.87][CONN][RXD] prevValue=1.000000
[1479239866.88][CONN][RXD] ain=1.000000
[1479239866.88][CONN][RXD]
```
This tells me that the value being read in by the AnalogIn object is always `1.00000`, which would be an error in the underlying library.

BlackstoneEngineering, 2016-11-15:
FYI this is an example of a successful run on a different platform, notice how the ain values are changing

```
[1479240121.11][CONN][RXD] outputs=0x1
[1479240121.13][CONN][RXD] prevValue=0.001724
[1479240121.15][CONN][RXD] ain=0.196109
[1479240121.15][CONN][RXD]
[1479240121.15][CONN][RXD] X=1
[1479240121.16][CONN][RXD] outputs=0x3
[1479240121.18][CONN][RXD] prevValue=0.196246
[1479240121.20][CONN][RXD] ain=0.398764
[1479240121.20][CONN][RXD]
[1479240121.20][CONN][RXD] X=2
[1479240121.21][CONN][RXD] outputs=0x7
[1479240121.23][CONN][RXD] prevValue=0.399084
[1479240121.25][CONN][RXD] ain=0.594736
[1479240121.25][CONN][RXD]
[1479240121.25][CONN][RXD] X=3
[1479240121.27][CONN][RXD] outputs=0xf
[1479240121.29][CONN][RXD] prevValue=0.594858
[1479240121.30][CONN][RXD] ain=0.797467
[1479240121.30][CONN][RXD]
[1479240121.30][CONN][RXD] X=4
[1479240121.32][CONN][RXD] outputs=0x1f
[1479240121.34][CONN][RXD] prevValue=0.797543
[1479240121.35][CONN][RXD] ain=0.995117
[1479240121.35][CONN][RXD]
```

ccli8, 2016-11-16:
Yes, NuMaker-PFM-M453 still fails. It has bug in analogin HAL. Anyway, the tests-api-analogin issue is fixed.

closed 2016-11-16 by ccli8