ARMmbed/ci-test-shield #13: PWM duty error rate > 5% in tests-api-pwm

ccli8, 2016-11-18:
**SHA**
6ca1b896edff34cacda2ed84e24455747262f295

**Description**
I meet this issue on NuMaker-PFM-NUC472 and NuMaker-PFM-M453 targets. About the function `tests-api-pwm` > `PWM_Duty_slave`, I have some comments:
1. Start PWM waveform recording after PWM configuration has finished. This could avoid recording premature PWM waveform.

    ```
    pwm.period((float)period_in_ms / 1000); // set PWM period
    pwm.write(duty_cycle_percent); // set duty cycle
    iin.rise(duty_cbfn_rise);
    iin.fall(duty_cbfn_fall);
    ```
1. Record for one or more PWM periods of time to record enough duty count.

    ```
    wait_ms((NUM_TESTS + 1)*period_in_ms);
    ```
    In `duty_cbfn_fall`, make sure correct duty count is recorded.
    ```
    if (duty_fall_count == NUM_TESTS) {
        return;
    }
    if(last_rise_time == 0){
        // do nothing
    } else{
        duty_fall_count ++;
        duty_running_count = duty_running_count + (duty_timer.read_ms() - last_rise_time);
    }
    ```

0xc0170, 2016-11-21:
@ccli8 Thanks for the feedback .Can you send patches ?

ccli8, 2016-11-21:
@0xc0170 My patch has been pointed out as above. Besides `PWM_DutyCycle_Test`, `PWM_Period_Test` may have similar weaknesses even though I haven't meet issues about it.

BlackstoneEngineering, 2016-11-22:
@0xc0170 at your discression, either update or mark for later. 

As it currently stands the error rate is less than 1%, so the 5% and 10% cases pass consistently.

ccli8, 2017-06-22:
The issue is majorly related to NUMAKER_PFM_NUC472 and NUMAKER_PFM_M453's PWM limit. I can fix its behavior to pass the test.

closed 2017-06-22 by ccli8