ARMmbed/ci-test-shield #22: I2CEeprom library update

LMESTM, 2017-01-06:
I have reported an issue in I2CEeprom library and proposed an update in the below PR:
https://developer.mbed.org/users/rhourahane/code/I2CEeprom/pull-request/1

But I haven't received an answer yet. 
An alternative is to use a fork of this library (mine or another one from ci-test team where the pull requests can be submitted for review, discussions and merges.

0xc0170, 2017-01-06:
It's been used in this repository. If it is not maintained anymore, we shall fork it and replace the reference here as well.

LMESTM, 2017-01-06:
@0xc0170 - so we're on the same page ?

0xc0170, 2017-01-11:
> @0xc0170 - so we're on the same page ?

Yes ! There was a pull request from geky to mbed-os, block device for I2C eeprom, that might be an answer. However it's a feature in progress currently for upcoming release

LMESTM, 2017-01-11:
@0xc0170  sounds good to me. In the meanwhile, you may consider reviewing my PR and using my fork of I2CC eeprom ?

LMESTM, 2017-03-03:
@0xc0170  any update on progress for this issue ?

BlackstoneEngineering, 2017-03-04:
@0xc0170 lgtm

0xc0170, 2017-03-04:
@LMESTM I shall look at this on Monday

0xc0170, 2017-03-06:
@LMESTM I forked I2C eeprom library to https://developer.mbed.org/users/mbed_official/code/I2CEeprom/ , please can you send a patch there? Let us know once done

LMESTM, 2017-03-06:
@0xc0170  - ok I tried to send the PR, I think it's here:
https://developer.mbed.org/users/mbed_official/code/I2CEeprom/pull-requests/?view=all
Not sure how I can check the diff though ...

0xc0170, 2017-03-06:
@LMESTM I left there a comment. I dont fully understand how it fixes the issue. As full write() should work, or should driver method be expanded to provide argumetn for repeated start? Just look at the PR and we can continue discussion there