ARMmbed/ci-test-shield #36: BOM Update

BlackstoneEngineering, 2017-03-07:
Suggested update: update I2C Flash chip to 512KB version (http://www.digikey.com/product-detail/en/microchip-technology/24LC512-I-SN/24LC512-I-SN-ND/2270915 ) so it can be used with the FS Tests. current one is too small. 

It is identical, should be drop in replacement. 

Priority: nice to have

BlackstoneEngineering, 2017-03-07:
@andcor02 @geky  - added based on conversation with geky, should be changed for next minor revision or major revision.

geky, 2017-05-25:
So, we've actually made several big improvements to the fatfs over in mbed OS. Soon we should be able to fit the fatfs onto the 32KB I2C chips currently on the CI shield (related pr https://github.com/ARMmbed/mbed-os/pull/3936). The eeprom upgrade is probably not necessary at this point.

For anything that needs a bigger size the addition of a spi flash chip is probably much more useful (https://github.com/ARMmbed/ci-test-shield/issues/37)

BlackstoneEngineering, 2017-05-29:
Still intending to add bigger flash size for other reasons (update, more expansive logging). 
@geky Should we add a FATFS test to the CI Test Shield list of tests? 

@senthilr @mray19027 to track