ARMmbed/ci-test-shield #37: Add SPI Flash chip

BlackstoneEngineering, 2017-03-07:
## Suggestion
Add SPI Flash chip to CI Test shiled

## Reason
Most deployments will use SPI Flash, not SD Cards. It makes sense to add a SPI Flash chip to the next version of the CI Test shield. 

## Implementation
Add 16MB SPI Flash chip to board, add CS Line to... not D0/D1.

BlackstoneEngineering, 2017-03-07:
@chris-styles @andcor02 - based on conversation with Chris. May be too late to implement in 2.0.x design, may wait till next major rev to implement, but should be added in future.

bridadan, 2017-03-31:
To avoid the flash wearing out, you might also consider using volatile storage, something closer to a RAM chip. Though this may have complications depending on the different test scenarios you have.