ARMmbed/ci-test-shield #39: Missing prerequisites for running ci test shield tests

maclobdell, 2017-03-08:
Need to add to the list of Pre-requisites

- git
- mercurial
- compiler such as GCC, ARM Compiler 5, or IAR

also python modules 

- mbed_ls 
- pyserial
 

For the python modules, how about a requirements file that can ensure they are installed?  Suggest to add a requirements.txt file. 
And add this command to the setup instructions:
`
pip install -r requirements.txt
`

0xc0170, 2017-03-09:
This is well know (to mbed OS) requirements. Shoudn't this be documented somewhere as part of mbed OS, and this repository just points to this documentation ?

BlackstoneEngineering, 2017-03-12:
@maclobdell we can add this to the readme, just to be extra explicit. 
@0xc0170 lets add a line to the readme that says it has the same requirements as mbed-os and then links to the requirements for mbed os?

0xc0170, 2017-03-14:
> @0xc0170 lets add a line to the readme that says it has the same requirements as mbed-os and then links to the requirements for mbed os?

PR sent

PR #41 (2017-03-14): Readme - add mbed-os requirements link
Fixes #39

@maclobdell @BlackstoneEngineering please review
closed 2017-03-14 by 0xc0170