ARMmbed/ci-test-shield #40: SD SPI test  not working with mbed-os/master

LMESTM, 2017-03-09:
Since FATFileSystem was introduced in mbed-os, the ci tests aren't compiling anymore because of redefinition of symbols. 

        Error: L6200E: Symbol ff_convert multiply defined (by ccsbcs.o and ccsbcs.o).
        Error: L6200E: Symbol ff_wtoupper multiply defined (by ccsbcs.o and ccsbcs.o).
        Error: L6200E: Symbol disk_initialize multiply defined (by FATFileSystem.o and diskio.o).
        Error: L6200E: Symbol disk_ioctl multiply defined (by FATFileSystem.o and diskio.o).
        Error: L6200E: Symbol disk_read multiply defined (by FATFileSystem.o and diskio.o).
        Error: L6200E: Symbol disk_status multiply defined (by FATFileSystem.o and diskio.o).
        Error: L6200E: Symbol disk_write multiply defined (by FATFileSystem.o and diskio.o).
        Error: L6200E: Symbol clust2sect(FATFS*, unsigned long) multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol get_fat(FATFS*, unsigned long) multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol put_fat(FATFS*, unsigned long, unsigned long) multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_chmod multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_close multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_closedir multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_getfree multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_lseek multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_mkdir multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_mkfs multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_mount multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_open multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_opendir multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_read multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_readdir multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_rename multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_stat multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_sync multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_truncate multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_unlink multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_utime multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol f_write multiply defined (by ff.o and ff.o).
        Error: L6200E: Symbol get_fattime multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::mkdir(const char*, int) multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::remove(const char*) multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::rename(const char*, const char*) multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::unmount() multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::~FATFileSystem__deallocating() multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::~FATFileSystem() multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Error: L6200E: Symbol FATFileSystem::~FATFileSystem__sub_object() multiply defined (by FATFileSystem.o and FATFileSystem.o).
        Not enough information to list the image map.
        Finished: 1 information, 0 warning and 37 error messages.

0xc0170, 2017-03-09:
Thanks for reporting. We are aware of this, once this repository gets update for the soon enough latest mbed OS (5.4), we update and fix those errors

cc @geky

LMESTM, 2017-03-09:
Note that I could not make it work by simply removing the code duplication. I am not sure whether the SDFileSystem lib is compatible with the mbed-os filesystem. I guess it may require to update ci-test to use SDBlockDevice from https://github.com/ARMmbed/sd-driver instead

0xc0170, 2017-03-14:
> I guess it may require to update ci-test to use SDBlockDevice from https://github.com/ARMmbed/sd-driver instead

Yes, the new filesystem should be used

jeromecoutant, 2017-03-15:
Fast workaround....:
rm -rf mbed-os/features/filesystem
:-)

BlackstoneEngineering, 2017-03-15:
We will update the CI Test Shield soon to use the FS in 5.4. Its a todo item for us.

0xc0170, 2017-03-20:
@LMESTM Please retest now, as 5.4.1 update PR was merged to master.

I'll close this issue, if any issue, feel free to reopen

closed 2017-03-20 by 0xc0170

LMESTM, 2017-03-20:
@0xc0170  I am not sure to understand. This is ci-test-shield repo that needs update, not mbed-os. Has ci-test-shield been updated already ? In which PR ?

0xc0170, 2017-03-20:
> @0xc0170 I am not sure to understand. This is ci-test-shield repo that needs update, not mbed-os. Has ci-test-shield been updated already ? In which PR ?

Reference: https://github.com/ARMmbed/ci-test-shield/pull/46

LMESTM, 2017-03-20:
@0xc0170   looks good :-) thx