ARMmbed/ci-test-shield #43: Avoid wrong AnalogOut test

jeromecoutant, 2017-03-16:
Hi

tests-api-analogout test is checking if device support DEVICE_ANALOGOUT,
but it supposes then, that this DAC pin is between A0 and A5...
which it is not always the case ...?

Maybe test should check first that AOUT pin is between A0 and A5 ?
Goal is just to avoid the "ERROR" result.

Thx

BlackstoneEngineering, 2017-03-17:
@jeromecoutant The current tests assume AOUT is mapped to A5. If the board has AOut on a different pin then it can be re-mapped in the mbed_app.json config file. For example the K64F board does not have an AnalogOut on A5,  the pin named DAC0_OUT is used. So to test this we fly wire DAC0_OUT to A5 and the test works.

closed 2017-03-17 by BlackstoneEngineering

jeromecoutant, 2017-03-17:
Quick answer....
What happened if DAC is connected to D13 ?

BlackstoneEngineering, 2017-03-17:
Then you can fly wire it to A5, though this will cause issues with the SPI tests as D13 is also the SPI_CLK pin.