ARMmbed/ci-test-shield #50: ci-test for ST platform in latest mbed OS 5.4

sunsmilearm, 2017-04-25:
I'm new helping a partner to do ci test for their new platform based on STM32F407. In previous mbed versions, they can pass most of the cases but in the latest version, the ci-test just blocked. But other examples such as blinky can work.

I find that in the new version of mbed OS, the ST driver folder is modified. I can repeat this issue on NUCELO_F401RE, I guess the new ST driver structure conflicts with ci-test.

import ci-test repo, use cmd: 
    mbed target auto
    mbed toolchain GCC_ARM
    mbed test -n tests-* --app-config .\mbed_app.json

After compile, the log stoped at:

```
mbedgt: greentea test automation tool ver. 1.2.2
mbedgt: test specification file 'C:\my\dev\mbed_5_0\oob_540\ci-test-new\BUILD\tests\NUCLEO_F401RE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\my\dev\mbed_5_0\oob_540\ci-test-new\BUILD\tests\NUCLEO_F401RE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'NUCLEO_F401RE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-api-analogin'
        test filtered in 'tests-api-businout'
        test filtered in 'tests-api-digitalio'
        test filtered in 'tests-api-i2c'
        test filtered in 'tests-api-interruptin'
        test filtered in 'tests-api-pwm'
        test filtered in 'tests-api-spi'
        test filtered in 'tests-assumptions-analogin'
        test filtered in 'tests-assumptions-digitalio'
        test filtered in 'tests-assumptions-i2c'
        test filtered in 'tests-assumptions-pwm'
        test filtered in 'tests-assumptions-pwmout'
        test filtered in 'tests-assumptions-spi'
mbedgt: running 13 tests for platform 'NUCLEO_F401RE' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
```

mray19027, 2017-04-25:
Can you add `-vv` when you run mbed test?
`mbed test -n tests-* --app-config .\mbed_app.json -vv`

And then paste the results here?

0xc0170, 2017-04-26:
@sunsmilearm please always provide details to reproduce this issue ! What is mbed-os sha? I assume you modified it to use latest master?

sunsmilearm, 2017-04-26:
@0xc0170 Nothing is modified

sunsmilearm, 2017-04-27:
Does anyone have update?

0xc0170, 2017-04-28:
> Does anyone have update?

I havent got a chance to reproduce.

cc @bcostm @adustm @jeromecoutant @LMESTM - you might know more to this one?

jeromecoutant, 2017-05-02:
Hi

Test with NUCLEO_F401RE, MBED set to version 5.4.4: there is no issue

mbed test -m NUCLEO_F401RE -t GCC_ARM -n tests* -vv

...
Build successes:
  * NUCLEO_F401RE::GCC_ARM::MBED-BUILD
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-ANALOGIN
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-BUSINOUT
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-DIGITALIO
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-I2C
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-INTERRUPTIN
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-PWM
  * NUCLEO_F401RE::GCC_ARM::TESTS-API-SPI
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-ANALOGIN
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-DIGITALIO
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-I2C
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-PWM
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-PWMOUT
  * NUCLEO_F401RE::GCC_ARM::TESTS-ASSUMPTIONS-SPI

mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: test specification file 'C:\github\ci-test-shield\BUILD\tests\NUCLEO_F401RE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\github\ci-test-shield\BUILD\tests\NUCLEO_F401RE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_F401RE | NUCLEO_F401RE[0]     | COM3        | F:          | 07200221053D64003A79F21D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'NUCLEO_F401RE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_F401RE | NUCLEO_F401RE[0]     | COM3:9600   | F:          | 07200221053D64003A79F21D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-api-analogin'
        test filtered in 'tests-api-businout'
        test filtered in 'tests-api-digitalio'
        test filtered in 'tests-api-i2c'
        test filtered in 'tests-api-interruptin'
        test filtered in 'tests-api-pwm'
        test filtered in 'tests-api-spi'
        test filtered in 'tests-assumptions-analogin'
        test filtered in 'tests-assumptions-digitalio'
        test filtered in 'tests-assumptions-i2c'
        test filtered in 'tests-assumptions-pwm'
        test filtered in 'tests-assumptions-pwmout'
        test filtered in 'tests-assumptions-spi'
mbedgt: running 13 tests for platform 'NUCLEO_F401RE' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_F401RE -p COM3:9600 -f "BUILD/tests/NUCLEO_F401RE/GCC_ARM/TESTS/assumptions/SPI/SPI.bin" -e "TESTS\host_tests" -d F: -C 4 -c shell -t 07200221053D64003A79F21D
mbedgt: mbed-host-test-runner: started
....
mbedgt: test suite report:
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                  | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-analogin          | OK     | 15.34              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-businout          | OK     | 28.4               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-digitalio         | OK     | 12.16              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-i2c               | OK     | 13.31              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-interruptin       | OK     | 12.94              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-pwm               | OK     | 175.94             | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-api-spi               | FAIL   | 14.42              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-analogin  | OK     | 25.26              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-digitalio | OK     | 12.97              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-i2c       | OK     | 10.67              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-pwm       | OK     | 12.57              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-pwmout    | OK     | 11.05              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-assumptions-spi       | OK     | 11.21              | shell       |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
...

sunsmilearm, 2017-05-02:
@jeromecoutant Thanks for your response! I will check this again and update later.

jeromecoutant, 2017-06-21:
Hi
Maybe we can close this issue ?
Thx

BlackstoneEngineering, 2017-06-22:
Closing issue.

closed 2017-06-22 by BlackstoneEngineering