ARMmbed/ci-test-shield #55: MBED_CONF_APP_DIO_X not well named for PWM tests

LMESTM, 2017-06-28:
I just noticed that MBED_CONF_APP_DIO_X have been added to be used for PWM tests and others.

MBED_CONF_APP_DIO_2 means D2, it makes no sense to re-redefine MBED_CONF_APP_DIO_2 to something else I think, ... clearly I have mis-understood something.

In case of PWM, what we need is to defined a pair PWM-OUT / PWM-IN 
where
  PWM-OUT is a valid IO for a pwm   (MBED_CONF_APP_PWM_0 used for this purpose)
  PWM-IN is the IO that is wired with a resistor to PWM-OUT IO on the CI shield 

so rather than MBED_CONF_APP_DIO_X, I would propose to use something like MBED_CONF_APP_PWM_0_INT 
MBED_CONF_APP_PWM_1_INT
MBED_CONF_APP_PWM_2_INT
MBED_CONF_APP_PWM_3_INT
meaning input interrupt to check PWM_0 

this would avoid dependency between PWM tests and BusInOut / InterruptIn tests

BlackstoneEngineering, 2017-06-28:
This is done so that boards that do not support arduino headers can still use the CI Test shield. I agree that this is not the best setup, but at the moment it is good enough. When we move to the level testing scheme as proposed in https://github.com/ARMmbed/ci-test-shield/pull/47 this will cease to be an issue as the tests will become self aware of what peripheral tests can be run on what pins. When this happens the mapping will simply say what MCU Pins are connected to which header.

LMESTM, 2017-06-28:
And what if I'd like to use D3 as PWM with an interrupt check on D2 input, instead of PWM on D2 with check on D3 ?  will that be possible ?