ARMmbed/ci-test-shield #56: Consider adding a time keeping device

bridadan, 2017-06-29:
In mbed OS testing, we currently struggle to have a trusted time keeping source to compare the hardware timers. We have been using the host PC as the trusted time source, but it isn't a real time system so the time can skew significantly by the time it is communicated with the device.

If we could have something like an RTC or a 555 timer circuit as a trusted time source, it would allow us to more accurately measure time and compare it with the device-reported time.

0xc0170, 2017-06-29:
We got RTC in HAL, would that be sufficient? Or you are proposing to have external RTC module, that we would compare against?

bridadan, 2017-06-29:
@0xc0170 yeah, something that we can trust with an externally versioned RTC library/implementation.

Also, not all boards have an RTC, but all boards do have timers that would be nice to verify!

BlackstoneEngineering, 2017-07-05:
Will take this into consideration for the next major revision of hardware! Thanks for the feedback! Will keep you in the loop going forward.