ARMmbed/ci-test-shield #58: Request to improve the PWM tests in order to support single edge trigger MCUs.

khj098765, 2017-07-11:
# << Request Information >>
## • Contents: Request to improve the PWM tests in order to support single edge trigger MCUs. 
## • Reasons

- Currently, single edge trigger MCUs cannot test PWM via CI test shield. 
`iin.rise(duty_cbfn_rise);`
`iin.fall(duty_cbfn_fall);`

- Not all the MCUs are dual edge triggerable. The single edge trigger MCUs only support either rise() or fall().

- Manual testing is theoretically possible. However, if it is not automated, you cannot check regression for future git submits. 

- Manual testing is practically impossible. In order to test, you have to check the precision with an oscilloscope or a logic analyzer, and you have to verify many samples to check the stability.

andrewc-arm, 2017-07-11:
Hi, team
FYI. @khj098765 is our communication channel and an engineer in WIZnet Inc., mbed Silicon partner. Please regard this issue with reasonable priority. Thanks,

BlackstoneEngineering, 2017-07-12:
Adding this as an enhancement. We will add this in the next round of enhancements to the tests. Currently targeting end of Q3 for the next release. 

For basic tests we will turn the existing single PWM test into 3 tests. One for Rise, one for Fall and one for Rise_Fall.

For Level testing we will propogate these tests into the level 1 tests as well as corner case testing. 

The lack of dual edge support on the Wiznet board should not be viewed as a 'failure', but it should be noted and the note should be put on the board page so users are made aware as well. We will make changes tot he CI Test Shield project to reflect this and make it more clear what is happeneing. 

@mray19027  for reference

mray19027, 2017-07-12:
The new folder structure will be as follows:
- Assumptions
- API
    |-- PWM_RISE
    |-- PWM_FALL
    |-- PWM_RISE_FALL
- Concurrent

PWM_RISE and PWM_FALL tests will only run 1 period with 1 duty cycle to avoid long test times
PWM_RISE_FALL will run the 3 designated periods and duty cycles

khj098765, 2017-07-12:
OK, I already done. 
I put on the board page so users are made aware as well.
https://developer.mbed.org/platforms/WIZwiki-W7500/
content is below
Notices : Due to low need, Dual Edge Triggered GPIO will not be supported. InterruptIn class’ rise() or fall() member function will override the previous callback setting.

is it right?

BlackstoneEngineering, 2017-07-13:
Might I suggest : 
```
<<warning title="PWM - single edge">>
PWM on this platform is single trigger. That means you can use either `pwm.rise()` or `pwm.fall()` but not both at the same time. If you try to use both then the software will only respect the last requested edge trigger. 
<</warning>>
```

mray19027, 2017-07-18:
@BlackstoneEngineering @khj098765 Added feature in PR #60. Can you verify that the 3 new tests satisfy the feature requested?

BlackstoneEngineering, 2017-07-20:
It looks like it will do the job. This should make it so that the Wiznet board passes the rise and fall tests but fails the rise_fall test. 

@khj098765 can you please test out this PR and verify on your hardware as we do not have any wiznet boards in our office.

closed 2017-08-07 by mray19027