ARMmbed/ci-test-shield #64: I2C-EEProm write failed because it requires a little deplay

soramame21, 2017-07-23:
`tests-api-i2c` got failed. 

## Log analysis
1. EEProm WR passed when length is 2 and 10 bytes, but failed when data length is 100 bytes at Case 9.
2. It means that a string (99 chars and a ending NULL) was written to I2C slave, right after read 100 bytes data from the same address, but the read string is different from the one written. even read string size is different from written string's.
3. and the failure happened when string size is bigger then page size (32 bytes).

**A. The log with more printf,  the failed point is marked by <<** 

> [1500793846.72][CONN][RXD] >>> Running case #5: 'I2C -  EEProm WR 2 Bytes'...
> [1500793846.77][CONN][INF] found KV pair in stream: {{__testcase_start;I2C -  EEProm WR 2 Bytes}}, queued...
> [1500793846.78][CONN][RXD]
> [1500793846.78][CONN][RXD] ****
> [1500793846.80][CONN][RXD]  Address = `5`
> [1500793846.81][CONN][RXD]  Len = `2`
> [1500793846.84][CONN][RXD]  Num Bytes Written = `2`
> [1500793846.87][CONN][RXD]  Num Bytes Read = `2`
> [1500793846.90][CONN][RXD]  Written String = `D`
> [1500793846.92][CONN][RXD]  Read String = `D`
> [1500793846.92][CONN][RXD] ****
> [1500793846.98][CONN][INF] found KV pair in stream: {{__testcase_finish;I2C -  EEProm WR 2 Bytes;1;0}}, queued...
> [1500793847.03][CONN][RXD] >>> 'I2C -  EEProm WR 2 Bytes': 1 passed, 0 failed
> [1500793847.03][CONN][RXD]
> [1500793847.09][CONN][RXD] >>> Running case #6: 'I2C -  EEProm 2nd WR 2 Bytes'...
> [1500793847.14][CONN][INF] found KV pair in stream: {{__testcase_start;I2C -  EEProm 2nd WR 2 Bytes}}, queued...
> [1500793847.15][CONN][RXD]
> [1500793847.16][CONN][RXD] ****
> [1500793847.17][CONN][RXD]  Address = `5`
> [1500793847.18][CONN][RXD]  Len = `2`
> [1500793847.21][CONN][RXD]  Num Bytes Written = `2`
> [1500793847.24][CONN][RXD]  Num Bytes Read = `2`
> [1500793847.27][CONN][RXD]  Written String = `P`
> [1500793847.29][CONN][RXD]  Read String = `P`
> [1500793847.30][CONN][RXD] ****
> [1500793847.35][CONN][INF] found KV pair in stream: {{__testcase_finish;I2C -  EEProm 2nd WR 2 Bytes;1;0}}, queued...
> [1500793847.41][CONN][RXD] >>> 'I2C -  EEProm 2nd WR 2 Bytes': 1 passed, 0 failed
> [1500793847.41][CONN][RXD]
> [1500793847.47][CONN][RXD] >>> Running case #7: 'I2C -  EEProm WR 10  Bytes'...
> [1500793847.52][CONN][INF] found KV pair in stream: {{__testcase_start;I2C -  EEProm WR 10  Bytes}}, queued...
> [1500793847.53][CONN][RXD]
> [1500793847.54][CONN][RXD] ****
> [1500793847.55][CONN][RXD]  Address = `100`
> [1500793847.57][CONN][RXD]  Len = `10`
> [1500793847.60][CONN][RXD]  Num Bytes Written = `10`
> [1500793847.62][CONN][RXD]  Num Bytes Read = `10`
> [1500793847.66][CONN][RXD]  Written String = `ABNYCLCBR`
> [1500793847.69][CONN][RXD]  Read String = `ABNYCLCBR`
> [1500793847.70][CONN][RXD] ****
> [1500793847.75][CONN][INF] found KV pair in stream: {{__testcase_finish;I2C -  EEProm WR 10  Bytes;1;0}}, queued...
> [1500793847.80][CONN][RXD] >>> 'I2C -  EEProm WR 10  Bytes': 1 passed, 0 failed
> [1500793847.81][CONN][RXD]
> [1500793847.87][CONN][RXD] >>> Running case #8: 'I2C -  EEProm 2nd WR 10  Bytes'...
> [1500793847.93][CONN][INF] found KV pair in stream: {{__testcase_start;I2C -  EEProm 2nd WR 10  Bytes}}, queued...
> [1500793847.94][CONN][RXD]
> [1500793847.94][CONN][RXD] ****
> [1500793847.96][CONN][RXD]  Address = `100`
> [1500793847.97][CONN][RXD]  Len = `10`
> [1500793848.00][CONN][RXD]  Num Bytes Written = `10`
> [1500793848.03][CONN][RXD]  Num Bytes Read = `10`
> [1500793848.06][CONN][RXD]  Written String = `EEHBSCLIV`
> [1500793848.10][CONN][RXD]  Read String = `EEHBSCLIV`
> [1500793848.10][CONN][RXD] ****
> [1500793848.16][CONN][INF] found KV pair in stream: {{__testcase_finish;I2C -  EEProm 2nd WR 10  Bytes;1;0}}, queued...
> [1500793848.22][CONN][RXD] >>> 'I2C -  EEProm 2nd WR 10  Bytes': 1 passed, 0 failed
> [1500793848.22][CONN][RXD]
> [1500793848.28][CONN][RXD] >>> Running case #9: 'I2C -  EEProm WR 100 Bytes'...
> [1500793848.33][CONN][INF] found KV pair in stream: {{__testcase_start;I2C -  EEProm WR 100 Bytes}}, queued...
> [1500793848.39][CONN][RXD] :75::FAIL: String Written != String Read     **<< failed 1st point! hit TEST_ASSERT_MESSAGE,  write string != read string, length: 100 bytes** 
> [1500793848.43][CONN][RXD] :76::FAIL: String Written != String Read
> [1500793848.65][CONN][RXD] :77::FAIL: Expected 'YRGCTGXMUZXVMVDVZVCZXGSBXUTRUKGRKQGXPIDXMXDSYMYKAMVCEHGRBWKQVFRTLWAGVRTJVQWBKZJOVPBMHNKFDEZXFWSVUVQ' Was 'YRGCTGXMUZXVMVDVZVCZXGSB@'. String read does not match the string written
[... 120 lines cut ...]

        // Write the bytes out one at a time to avoid having to copy them to
        // another buffer.        
        for (int count = 0; count != toWrite; ++count) {
            if (m_i2c.write(*page) == 0) {
            // Write failed to return bytes written so far.
                return size - left;
            }
            ++page;
        }
        
        // Stop the transaction now we've completed the page
        // write.
        m_i2c.stop();
        
        waitForWrite();
        
        // Update the counters with the amount we've just written
        left -= toWrite;
        address += toWrite;
    }
    
    return size;
}
````

BlackstoneEngineering, 2017-07-25:
@soramame21 , please specify what platform you are seeing this failure on. Also please check that you are using the latest tests, please provide a `git log --oneline -n1` so we can confirm your version

soramame21, 2017-07-27:
@BlackstoneEngineering This is a pre-release platform working with 5v not 3.3v. the ci-test-shield is modified specially. I'm using the latest version of ci-test-shield. 

> dfce4e2 (HEAD -> master, origin/master, origin/HEAD) Add repository structure to readme

Since the working environment is different from others, I think we may not need to make further investigation on it. I'd like to close this issue. What is your comment or suggestion?

soramame21, 2017-07-27:
@BlackstoneEngineering I close it now. if it's necessary to go deeper, we can reopen it.

closed 2017-07-27 by soramame21