ARMmbed/ci-test-shield #89: I2CEeprom compilation failed with mbed-os master

jeromecoutant, 2019-06-10:
Hi

Due to new MBED compilation options, I2CEeprom can't be compiled any more...

````
[Error] I2CEeprom.cpp@32,23: non-constant-expression cannot be narrowed from type 'std::size_t' (aka 'unsigned int') to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@32,39: non-constant-expression cannot be narrowed from type 'unsigned int' to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@47,23: non-constant-expression cannot be narrowed from type 'std::size_t' (aka 'unsigned int') to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@47,39: non-constant-expression cannot be narrowed from type 'unsigned int' to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@62,23: non-constant-expression cannot be narrowed from type 'std::size_t' (aka 'unsigned int') to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@62,39: non-constant-expression cannot be narrowed from type 'unsigned int' to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@101,27: non-constant-expression cannot be narrowed from type 'std::size_t' (aka 'unsigned int') to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@101,43: non-constant-expression cannot be narrowed from type 'unsigned int' to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@161,27: non-constant-expression cannot be narrowed from type 'std::size_t' (aka 'unsigned int') to 'char' in initializer list [-Wc++11-narrowing]
[Error] I2CEeprom.cpp@161,43: non-constant-expression cannot be narrowed from type 'unsigned int' to 'char' in initializer list [-Wc++11-narrowing]
````

jeromecoutant, 2019-06-10:
@MarceloSalazar @c1728p9

MarceloSalazar, 2019-06-12:
@jeromecoutant thanks for reporting this.
I confirm I see this issue when trying 5.13.0-rc2 but not when using 5.12.4

@ARMmbed/mbed-os-maintainers can you please have a look?
It looks some flags have been added recently which break compatibility applications/tests such this one.

adbridge, 2019-06-12:
@MarceloSalazar is this an officially supported compatibility test? I haven't seen this before and I suspect it is not run in CI ?

adbridge, 2019-06-12:
The only obvious change I can see which might impact is this one https://github.com/ARMmbed/mbed-os/pull/10427 , though I don't know what the impact is. I will ask @kjbracey-arm  could you take a look please ?

kjbracey-arm, 2019-06-13:
This would be https://os.mbed.com/users/mbed_official/code/I2CEeprom/ - discussed it with @0xc0170 , but neither of us have taken action. I'm not sure who's supposed to be responsible for that.

The code is not C++11 compatible - there's a new C++11 rule that you can't have narrowing conversions in initialiser lists. Conversions that might change the value are barred unless the compiler can see that they're a constant expression that doesn't.

This rule only applies to `{}` initialiser lists, not any other parameter passing or assignment.

So:

      uint32_t x;
      void foo(uint8_t x);
      foo(x); // OK
      foo(x & 0xFF); // OK
      uint8_t array1[] = { x }; // not OK in C++11
      uint8_t array2[] = { x & 0xFF }; // not OK in C++11
      uint8_t array3[] = { 0x100 }; // not OK in C++11
      uint8_t array4[] = { 0x100 & 0xFF } ; // OK
      uint8_t array5[] = { static_cast<uint8_t>(x) }; // OK
      uint8_t array6[] = { (uint8_t) x }; // OK
      uint32_t value1 = { -1 } ; // not OK in C++11
      uint32_t value2 = -1; // OK

0xc0170, 2019-06-13:
> This would be https://os.mbed.com/users/mbed_official/code/I2CEeprom/ - discussed it with @0xc0170 , but neither of us have taken action. I'm not sure who's supposed to be responsible for that.

This should be fixed. 
@donatieng Would be anyone from HAL team to able to help to fix this repository ?

MarceloSalazar, 2019-06-21:
@donatieng bump

MarceloSalazar, 2019-06-27:
Internal reference: IOTHAL-508

jeromecoutant, 2019-07-04:
patch: https://github.com/ARMmbed/ci-test-shield/pull/91

jeromecoutant, 2019-07-04:
@mtomczykmobica

closed 2019-07-04 by jeromecoutant