ARMmbed/ci-test-shield #93: Correct deprecated API

jeromecoutant, 2019-12-02:
Lots of deprecated API are used with ci-test-shield

- wait_ms
- wait
- gettid

We need to use this shield even after mbed-os 5.14

Thx

@MarceloSalazar 
@LMESTM 
@bulislaw

bulislaw, 2019-12-04:
@jamesbeyond please have a look

jamesbeyond, 2019-12-04:
you mean in https://github.com/ARMmbed/ci-test-shield
@jeromecoutant ?

jeromecoutant, 2019-12-04:
Yes, + I2CEeprom + LM75B

jamesbeyond, 2019-12-04:
Despite we announced deprecation of those APIs, I'd say we haven't done a good job in terms of deprecating those APIs from our end. Some of our code still uses them. We have a task for cleaning up our deprecated APIs from mbed-os code base, I will make sure those repos will get updated along with the mbed-os

jeromecoutant, 2020-03-05:
Part of correction in #96

jamesbeyond, 2020-03-30:
I believe the deprecated API from this repo had been updated. closing this ticket

closed 2020-03-30 by jamesbeyond